SilentNightx/VitaDockPlus #6: Not working with Pi 4b 8GB

ss9918, 2020-12-25:
Tried this image with a Pi 4b 8gb model and just getting the 4 green flashes no boot, the Pi is good have tried with other images

naifmeh, 2021-01-13:
Just tried on that same Pi, not working for me as well. No boot at all.

ss9918, 2021-01-17:
Any solutions for Pi 4b 8GB?

BeauJade, 2021-01-20:
Same issue here

PasqualeMontone, 2021-02-14:
Any solution?

PasqualeMontone, 2021-02-18:
Fixed with 1.2

HelyeFab, 2021-03-09:
@PasqualeMontone 
Hi there, 
any chance you could share you managed to get it working on PI4 8gb. I have tried with version 1.2 and several sd card but to no avail. Any idea would be great!
CHEERS

PasqualeMontone, 2021-03-12:
@HelyeFab 
On my p4 8gb the v1.2 works without any changes.

dantesquivel, 2022-04-11:
I have also attempted v1.2 on RP4 and get a prompt to update the software. I was using the 
GeeekPi DeskPi Lite to output the signal (which has a "siter-board" with a power button and regular sized hdmi that connect through the usbc and micro hdmi), but I don't believe that's the problem since there was a signal output (hence the error promp) and the sd card is inserted in the RP4, not the "sister-board"

SilentNightx, 2022-07-20:
Resolved in latest image.

closed 2022-07-20 by SilentNightx