SilentNightx/VitaDockPlus #21: Bluetooth audio issues

Stonners23, 2022-07-23:
Thanks for the updated version - loving all the updates.  Using a rasp pi 3b and can link to bluetooth, but no audio comes through to TV.  Have amended audio settings to HDMI out, but Vita device options defaults to off, no matter how many times I change to one of the 3 options there.  Help?

Snake-Programerlod, 2022-07-23:
I have the same issue that @Stonners23, my raspberry pi is the same model to, when select High Fidelity Capture, if is paried to vita, the conection was aborted, and when pairing again the High Fidelity change to off, i hope could you help us

SilentNightx, 2022-07-24:
Please try the legacy image I just uploaded and report back. Also please provide your specific model and revision using the instructions on the GitHub readme.

Stonners23, 2022-07-24:
Great stuff thanks.  Installed and all seems to be working as it should :)

SilentNightx, 2022-07-24:
Due to these issues I've rolled back some changes. The image that caused this issue is now labeled experimental and the legacy image is the new main image.

closed 2022-07-24 by SilentNightx