SilentNightx/VitaDockPlus #24: RESOLVED [but see comments] No audio through HDMI even with Bluetooth correctly paired

dantesquivel, 2022-08-10:
The below solutions only works for the VitaDock Plus 2.0 version that was originally released 7/21/2022 (the one that had display settings available and could select audio source profiles), the current software does not work with audio through HDMI.

I have managed to resolve the lack of audio through HDMI following a few steps (again, for the original version of the VD+ 2.0 which is no longer posted):

- connecting the VD+  to the VITA once, then from the VITA disconnecting and DELETING the VD+
- Ensure that the VD+, when you right click on the audio profiles that HDMI is an option and it has a green checkmark, if HDMI does not appear, you have to go into profiles and manually add and apply. 
- from the VD+ menu, selecting remove the VITA (but DON'T confirm in the dialog box that comes up) and restart the VD+
- For some reason, this triggers a VD+ boot screen with the PS logo and three squares lined up horizontally underneath
- Restart the VITA
- Once the VD+ comes up again, you should have an instance of PlayStation Vita in the VD+'s BT settings with a red dot x (not connected)
- Make the VD+ BT discoverable
- On the VITA, once again connect to the VD+, BT pairing will show a code on both VITA and VD+ and you have to confirm both
- [Note that this is the step where the current VD+ 2.0 does not work, because it does not allow you to pair again since you already have an instance of the VITA in the VD+ BT devices. However with the original release...]
- At this point you will usually have sound once connected, note that in the VD+ there will be some dialog that comes up about 'trying to connect to bt device" - don't cancel that, if you still don't have sound, reboot the VD+
- You should have sound through HDMI now, note that you cannot control the BT volume through the VITA when correctly pushing audio through HDMI, so if you have BT paired, and you can control BT audio through the VITA, something went wrong with the above steps, start over by fully removing each BT device from the other
- You only have to set this up once per TV and VD+, afterward you just connect the VitaDock + in the VITA and you get audio

@SilentNightx - please consider posting the original VD+ software, released on 7/21/2022, which had the display settings available through the menu and let you manage the audio profiles from right clicking the speaker, as both of those are needed to get this to work.

dango90ec, 2022-08-12:
@SilentNightx I have the same problem. Could you post the original VD+ software, released on 7/21/2022 please? I want to try @dantesquivel solution. Thanks

dantesquivel, 2022-08-12:
Just wanted to add another strange issue. I'm pushing the Vita HDMI signal through an AV receiver. The receiver will identify an HDMI signal with a little HDMI label. When I rebooted the VITA this HDMI signal will flash for a few moments, then it disappears. If I don't manage to connect the vita to vd+ through Bluetooth while this HDMI signal is flashing, I need to reboot the VD+. If I am able to, it stops flashing and stays.

Now I'm wondering if that whole process I described above is simple a convoluted way of having that HDMI signal "locked" quickly because the dock already has an instance of the Vita? Dunno
[image]

SilentNightx, 2022-08-13:
I'll read through this issue when I have time but for now know that the "Experimental" release IS the original release of 2.0.

dantesquivel, 2022-08-14:
Ok great, thank you, I'll check it out!

dantesquivel, 2022-09-02:
I was really hoping my long winded method described above would be a 1 time thing, and then audio through HDMI would just work by connecting the Bluetooth. Sadly, any time I want to use the VitaDock+ I have to go through this same song and dance to get the HDMI signal locked in. Usually the HDMI indicator just blinks a few times and then disappears, there will be video bit no sound unless I troubleshoot again.

@SilentNightx any thoughts here? I can attest yo HDMI working through Bluetooth, it's just extremely unreliable.

SilentNightx, 2023-06-16:
Should be fixed in V2.5.

closed 2023-06-16 by SilentNightx

roiKosmic, 2024-03-30:
I have same issue. Vita welle connected in bluetooth to the dock (raspberry Pi3 + ) but no sound. I have the 2.5 img
Perhaps you could guide me to the conf file to update ?