SilentNightx/VitaDockPlus #28: [ENHANCEMENT] Update Boot animation

Girotin, 2022-10-30:
Hello, I don't even own a raspberry PI to test this, but I like this idea so much that I made small custom fmvs for the Bootup of VitaDock

______________________________________________________________________________________________________________________________________________
https://user-images.githubusercontent.com/45373992/198902854-4a9d0812-152e-41d8-b927-922210d0eee2.mp4

______________________________________________________________________________________________________________________________________________

https://user-images.githubusercontent.com/45373992/198902897-cd34fa4c-d841-4046-a6c1-9914f481d861.mp4

______________________________________________________________________________________________________________________________________________
Do you think one of these could make it?

SilentNightx, 2023-06-16:
These are very cool. The answer is maybe if it can be done as Plymouth splash screen. Unfortunately it's not something I'm familiar with.

Gdkdns, 2023-07-04:
> Hello, I don't even own a raspberry PI to test this, but I like this idea so much that I made small custom fmvs for the Bootup of VitaDock [...]

Yes I like this idea. I think it will make it. I just built my raspberry pie 4 works perfectly.

francoisjacques, 2024-08-03:
It's beautiful. [This repo](https://github.com/PROxZIMA/proxzima-plymouth) demonstrates an animation through frame by frame animation.