SilentNightx/VitaDockPlus #30: issue with MEGA download quota

harmadiwibowo, 2022-12-02:
Hi, can you give us download option that can be downloaded free without quota, since MEGA just give us 3GB quota and we must wait for 5 hours ++ to wait until the quota is back to normal. Thank you.

andypoo79, 2023-01-12:
Hi, is it possible to provide another download link please?

I would love to try this project ut cannot access it via Mega 

many thanks

pslawx, 2023-01-14:
Hi, I also would love to try it, but MEGA with free quota usage is not the ideal place, could you please upload on Google Drive or some other place that let us to download? Or if possible give us a tutorial showing how to build this image on our own machines. Thanks a lot!

SilentNightx, 2023-06-16:
Added another download option to the new release. If anyone wants to help host a torrent let me know.

closed 2023-06-16 by SilentNightx