SilentNightx/VitaDockPlus #31: [Feature Request] Support for Raspberry Pi 4 Model B Rev 1.4

blackson44, 2023-01-20:
Hi, I tried both the VitaDock+ 2.0 & the experimental version, both had no response when I plug in my PS Vita. However, when I plug in my PS Vita into the PC, it works fine with the OBS studio.

SilentNightx, 2023-06-16:
Should be fixed in V2.5.

closed 2023-06-16 by SilentNightx

blackson44, 2023-06-23:
Thank you for your reply. I tried the V2.5 but still no response. Maybe there is some issue on my PSVita since I could not get the ucdc_uvc driver install with the Autoplugin 2, an error occurred and it reboot. So I tried to install it manually by putting the file in the ur0:tai folder and adding the command line in the config.txt file. Then, I can stream the video to the PC with the OBS Studio. However, there is no response on the PSVita Dock. Below are the plugins on my Vita, maybe one of those conflict with the ucdc_uvc:
storagermgr.skprx
nonpdrm.skprx
reFOOD.skprx
repatch.skprx
ucdc_uvc.skprx
nopsmdrm.skprx
ds4vita.skprx