SilentNightx/VitaDockPlus #37: vidadockplus in raspberry pi 5?

balderofmidgard, 2023-12-17:
Hi, I would like to know if VitaDockPlus works in a Raspberry pi 5. Thanks.

samshosho, 2024-02-23:
doesn't work unfortunately.
it keeps outputting board info, usb info and never gets into booting the OS.
although it has a message saying 'The installed operating system (OS) does not indicate support for Raspberry pi 5"
then another output that says,
"Update the OS or set os_check=0 in the config.txt to sip this check"

I don't know how to change the OS, but i did change the config.txt and added the line, however after adding it, nothing shows on the screen at all, not even the information it was giving before.

hope someone is able to fix this.

SilentNightx, 2024-11-07:
Pi 5 doesn't support bullseye so getting it to work requires a major version upgrade to bookworm. This is something I'm attempting but I'm currently stuck on an issue where mpv runs at <1fps when in fullscreen mode.