SilentNightx/VitaDockPlus #42: Audio Issue - Rapberry PI 3 + BT

AlexPaterson77, 2024-04-05:
My Psvita dock+ 2.5 installation is on a Raspberry PI 3, connected to the original docking station by SONY via USB and to my TV using an HDMI cable.

It works perfectly with video and also connects to my dualshock 3 with the ds34Vita by MERLev plugin. 
I am having issues when connecting my vita to raspberry 3 via bluetooth in order to pass audio. The stream is perfect, through hdmi, when the console is not on my dock. 
Though: whenever i plug the vita onto my dock or USB cable (i tried two different cables, one original, the other not, with and without original dock) audio stops. I tried turning up the volume but it simply doesn’t get out, neither from the tv nor from the vita.

VicMuuh, 2024-04-18:
It's exactly the same for me. Raspberry Pi 3 Model B Rev 1.2. 

Any advice?

SilentNightx, 2024-11-07:
Unable to replicate but I don't have the Pi 3.

AlexPaterson77, 2024-11-17:
Just tried version 2.6 of VitaDockPlus

On version 2.5, I could pair my vita with vitadock and stream audio through HDMI if the console was not connected to the dock.
On version 2.6, I can pair my vita with vitadock, but the audio streaming does not get out of HDMI, either when the console is not on the dock and when it is on the dock. 

In short: with 2.6, the console I connected via Bluetooth to my raspberry 3 starts being completely silent after the pairing, as if the stream is routed to something else, though no sound gets out of my TV.

SilentNightx, 2025-01-29:
With two reports I've decided to put it down as a potential issue with the image on Pi 3. If it works for anyone or if anyone finds a solution please update this issue.

Pandatective, 2025-11-15:
I was curious about the Vita Dock and since I have a RPI 3 around, I gave it a try. I have tested 2.5 and 2.6 and I couldn't get the BT audio working. However, I tried the "Current" 2.0 build and BT audio did work with this one, except it's cutting out from time to time.

I'm aware it has been a while since the last message in the discussion, but I was wondering if there may be other things I could check, or it would be more preferable to use a different RPI model or use AUX instead?

AlexPaterson77, 2025-11-16:
So, this means it’s actually an issue with Vita Dick Plus and not something related to the plugin on the ps vita. Cool to know.
Will try 2.0 as soon as I’m home

Pandatective, 2025-11-18:
To be a bit more specific, it works with "Current" 2.0 build, but not with the "Experimental" one, which the latter's main difference,  from my understanding, is based on Raspberry Pi OS Bullseye instead of Buster.

Pandatective, 2025-11-23:
Sorry I took so long to provide an update; Was preoccupied with other stuff.

Turns out I didn't have to look too far. Whenever I try a build based on Bullseye, I do a right-click on the volume icon in the taskbar and then notice the output is set to  "AV Jack" instead of "HDMI" by default.

The BT audio is now working after switching it on "HDMI", but I'm noticing a delay of a second for the audio. I could blame this on the RPi3, but the delay was far less noticeable in the 2.0 "current" build despite stuttering for a bit, so I'm not sure if there's adjustments that could be done.

AlexPaterson77, 2025-11-23:
Hi, I had already tried this solution back in time: it works until you put the psvita in the dock. If you connect it to the raspberry via usb, audio stops working.

Pandatective, 2025-11-23:
On this part, I do not know. 

For me, the audio works via Bluetooth before and after plugging it on the USB and it has been running for several minutes. Speaking of the delay, I left it running a bit longer and the delay is gone, as per noted in Readme.

Pandatective, 2025-11-25:
udcd_uvc_oled_off.skprx from https://github.com/xerpi/vita-udcd-uvc/releases/tag/1.7