SilentNightx/VitaDockPlus #49: Ways to get analog audio instead of bluetooth

guillemdc, 2025-01-09:
Hi,

I've got my pi zero w2 already working with bluetooth, but I wanted to test and compare the sound and lag coming directly from the jack.

Doing some research about usb dongles -as you suggest in the readme- most of them only have a mono input and it's impedance is more suitable to mics than it should be for a line/headphones input.

However, I came across this thing: [Codec Zero](https://www.raspberrypi.com/products/codec-zero/ )

Do you think it'd work?

SilentNightx, 2025-01-29:
I was originally looking for something like that when writing the readme with no luck, It'll probably work with some configuration but I'm unsure about if the AUX configuration script works with GPIO devices since I didn't write the AUX script. If anyone tries it please update us.

guillemdc, 2025-02-17:
Guys, thanks a lot for the answers. I think I'm going to try to develop my own pcb to attach it to a raspberry pi zero 2W. I want to embed in that pcb maybe a pcm2902 and have one side with a male stereo mini jack (that would go to the vita). Pass that through a pcm2902 (maybe, if these dongles work) and maybe isolate that with an ADuM3160 and that to a male micro usb to the pi.

On the other side, proprietary vita1000 connector to the other usb just wiring the +d to -2. Also, that'd allow me to put a USB-C to power the vita and an internal rewiring from a male micro-HDMI (that'd go into the pi) to a female full size HDMI exposed to the outside world. In theory it all should work, but I'm going to try to set it up first with dongles and adapters and, if it all works, I'll go for the full blown single pcb project :)

SourceK78, 2025-07-10:
I was successful in analog input using Audio Injector Zero sound card.

Some GPIOs conflict, so I edited [vitadock.conf](home/pi/vitadock.conf).

LEFT_KEY_GPIO=25
RIGHT_KEY_GPIO=24
DOWN_KEY_GPIO=23

https://pinout.xyz/pinout/audio_injector_zero_sound_card

Also, the physical power button (GPIO 3) conflicts with the Audio Injector's I2C SCL, so change the pin number in [/home/pi/pi-power-button/listen-for-shutdown.py](home/pi/pi-power-button/listen-for-shutdown.py) to an available pin, or uninstall pi-power-button.

Only wake-up works by connecting the button to GPIO 3.

I don't know which GPIO the Codec Zero uses, but I think it may work if you find the conflicting GPIO and set it the same way.

One last thing, as mentioned in the README of this repository, a ground loop isolator like the one shown in the photo is used to reduce noise.

Until then, have a good VitaDock+ life.

[image]

guillemdc, 2025-07-11:
Well, that's pretty neat!! It's a passive isolator, right? no power nor buffers. In my experience (I've done a few for guitar pedals) if they're not buffered (which would require a dual opamp, or a couple of JFETs as well as some resistors and getting external power) they tend to lose quite some quality. How does it sound?

SourceK78, 2025-07-11:
I don't have good enough ears to judge whether the sound quality is good, so this is just my personal opinion, but I think it's good enough for playing games.

I've attached an audio recording on my PC for reference.

The audio appears to be at a high rate because it is input to the PC via an S/PDIF connection via an HDMI audio separator, but it is actually a recording of analog audio.

[vitadocplus_rec.zip](https://github.com/user-attachments/files/21182035/vitadocplus_rec.zip)